mart1nro/joycontrol #1: dbus module not found

danybeam, 2020-02-21:
Hi :)
I'm trying to run your project on my Pi 3B and I'm having trouble importing dbus
I've tried installing dbus, python3-dbus and python3-gi (a requirement for dbus aparently) and none seem to work

could you share how you installed it please?

mart1nro, 2020-02-21:
The name of the package is dbus-python.

You should be able to install it using
`sudo apt install python3-dbus`
but I forgot and don't have access to my Pi at the moment. At least that's the package I'm using on my Ubuntu laptop and the package list of Raspian Buster suggests that it also exists there.

I'm going to update the README once I'm sure. Please message, if it worked. Also I'm planning to make a list of issues soon.

Just as a heads up, I have the problem where the Bluetooth connection randomly drops after a while using the Raspberry Pi 4B. When using my laptop it's fine, so it might be a hardware issue. If you run into the same problem, try using a different Bluetooth adapter.

danybeam, 2020-02-21:
It worked like a charm in my pi3B
the root of the issue was that I was using Raspbian jessie instead of buster which caused a lot of issues with python custom installation and stuff

for any future user *YOU DO HAVE TO INSTALL python3-dbus* 

I was just impatient and started doing stuff wrong

closed 2020-02-21 by danybeam