mart1nro/joycontrol #3: IndexError

SuperPichu, 2020-02-26:
I am seeing an index error running either run_test_controller_buttons.py or run_controller_cli.py on Debian 10 Buster
It connects, sends one button press correctly then disconnects from the switch and gives me this:
`[09:51:30] joycontrol.protocol callback::270 ERROR - list index out of range
Traceback (most recent call last):
  File "/home/chris/dev/joycontrol/joycontrol/protocol.py", line 268, in callback
    future.result()
  File "/home/chris/dev/joycontrol/joycontrol/protocol.py", line 99, in input_report_mode_0x30
    report = OutputReport(list(data))
  File "/home/chris/dev/joycontrol/joycontrol/report.py", line 219, in __init__
    if data[0] != 0xA2:
IndexError: list index out of range`

mart1nro, 2020-03-08:
The index error occurs if no data is received, most likely due to a disconnect. I added an error message now.

How consistent does the error occur? I'm having disconnect issues using my Raspberry Pi 4B, but it happens randomly and usually after several minutes. 

Anyway try a different bluetooth adapter and see if it helps.

SuperPichu, 2020-03-25:
It seems to just start happening and not stop, currently it appears as soon we pass Set_Player_Lights every time.

SuperPichu, 2020-03-25:
Yup it was my adapter. Oddly enough the builtin ones on my desktop and my laptop don't work but the $10 usb one does...... Sure why not?

I'm marking this closed.

closed 2020-03-25 by SuperPichu