mart1nro/joycontrol #4: stick_hold is note defined

weslcurt, 2020-03-15:
Is there a way to emulate inputs from joysticks? When I attempt to create a test that includes a stick_hold event I get an error that it's not defines. Which makes sense as I don't know how to define which stick or how to tell it where to go.

mart1nro, 2020-03-15:
The ControllerState class defines l_stick_state and r_stick_state, depending on the controller. in the StickState you can set controller values with set_v and set_h respectively. Also methods to set the stick to one of 4 directions exist.

One thing to note is, that sticks require calibration data defining the limit and center positions.
The Switch reads them from the spi flash of the controller. To apply the calibration, the flash memory must be emulated. The current solution is to create a memory dump of a real controller with the dump_spi_flash.py script (The script only works with joycons. Probably only needs minimal adaptions to make it work with the Pro Controller, but I don't have one and no way to test it).

I haven't much used the sticks yet, but they are implemented for testing purposes in the command line interface. It can be run with run_controller_cli.py and you can see a bit how they work there.

weslcurt, 2020-03-15:
Thanks a ton! This is super helpful and cool project to boot.

weslcurt, 2020-03-17:
Side note, if I were to have access to a PRO Controller, is VENDOR_ID and PRODUCT_ID the values you would need to implement in the dump_spi_flash.py (or possibly a new file)?

mart1nro, 2020-03-17:
The r_stick and l_stick command is for the button, sorry for the confusion. I should add some more comments in the file.

To change stick positions the command is
`stick <side> <direction>`
For example if you want to change the right stick to the up position, you write
`stick r up`
If you want to set the values directly, you can write
`stick r h <value>`
for the horizontal value and
`stick r v <value>`
for the vertical value.
Setting values directly doesn't require calibration data.

The VENDOR_ID of the Pro Controller is probably the same.
It should work, if you add the PRODUCT_ID of the Pro Controller in the check of line 154. You can find out the PRODUCT_ID if you just do a print(device) before that check.

If you have access to a Pro Controller and it works it'd be great if you submit a pull request.
Also I would really appreciate if you could send me a dump per mail. My mail address is written in the setup.py of the package.

closed 2020-03-19 by weslcurt