mart1nro/joycontrol #5: Raspberry Pi 3B issue update

danybeam, 2020-03-17:
to add to the list of known issues, raspberry pi 3B fails around the 15 minute mark (guesstimate)
<End Of Message>

danybeam, 2020-03-17:
[this](https://raspberrypi.stackexchange.com/questions/28578/python-script-stops-running-after-about-3-hours) seems like a potential fix
@mart1nro if you've tried it I would love to hear how it went
else I'll be trying it tomorrow morning and report back :)

mart1nro, 2020-03-17:
Hey, I'm not entirely sure what the issue is. Is it similar to the link you posted and the script stops working completely?

If you're referring to the Bluetooth disconnect issue I was mentioning, the script doesn't crash without error output.

Anyway, I did some more digging about the Bluetooth issue and apparently there was a bug fixed recently (see: https://github.com/raspberrypi/linux/issues/2832).
So I updated the firmware with
`sudo rpi-update`

It sees to be working now. Currently I have a controller script running for one and a half hours on my Raspberry Pi 4B and no disconnects so far.

danybeam, 2020-03-17:
Sorry I didn't write it fully conscious haha
The issue I'm having is that the script randomly stops after 15~30 minutes
I don't get any errors, just stops as if by ctrl+C I don't know what causes it though

I just ran a test and it definitely ran longer than usual using the command in the link but other than that I don't know what happened.
I'll update the firmware and will keep investigating on my end

danybeam, 2020-03-17:
As of time of writing the solution seems to have been running  
`sudo -b python3 InfiniteScript.py` 
it creates a root level process and has been going strong for 1hr 50m (properly counted this time)  
If it's still running tomorrow morning I will mark the issue as closed and edit the question to quickly surface the solution for the pi 3B

closed 2020-03-18 by danybeam