mart1nro/joycontrol #6: compatibility with other bluetooth application

LaolongVincent, 2020-03-19:
Thanks for your project. I have successed use a Pi as and Pro Controller. Furtherfore, I am trying to use the pi as converter for connecting a xbox controller with NS. If xbox controller connecting pi with usb, and using your code, it can work well. Then, I used the xpadneo driver which make xbox controller connecting pi via Bluetooth，the xbox controller /dev/input/js0 have been recognized. In this case, when I go your code again, the code can run, but it will appear some error when pair with NS. 
So I ask for advice. Thanks

mart1nro, 2020-03-20:
Currently there is a bluetooth restart required to set up for the connection with the Switch. This likely causes a disconnect with the xbox controller. It could work if you connect the xbox controller after the pi is waiting for a connection with the Switch.

LaolongVincent, 2020-03-26:
thanks for your advice. Now it can work. I will try the stability of that. thanks again

closed 2020-03-26 by LaolongVincent