mart1nro/joycontrol #7: Is it possible to emulate multiple joycon instances on a single machine at the same time?

gahoo, 2020-03-19:
This project is marvlous. But I wonder is it possible to emulate multiple joycon instances on a single machine at the same time?

mart1nro, 2020-03-20:
The Bluetooth ports for the controller are hard coded and so far as know cannot be changed. It would be possible to use another Bluetooth adapter for a second controller, but it's currently not supported in the code.

mart1nro, 2020-03-30:
Hey, I added support for multiple Bluetooth adapters now, so multiple controller instances can be emulated.

gahoo, 2020-04-08:
Wow, that's really nice. I'll give it shot when I got another Bluetooth adapter.

mart1nro, 2020-04-20:
I'm closing this for now. Feel free to reopen, if you have any more issues.

closed 2020-04-20 by mart1nro