mart1nro/joycontrol #8: Incompatibility with bluetooth "input" plugin

mart1nro, 2020-03-21:
The Switch can not connect to the ports of the emulated controller, if the "input" plugin in Bluez is enabled.

However, other Bluetooth input devices can't be connected, if the "input" plugin is disabled.
The currently implemented workaround is a restart of the Bluetooth stack, after sockets for the emulated controller are registered. If you have a similar use case like in issue #6, you need to reconnect your device after that restart.

If you don't need to connect any other Bluetooth input devices and want to avoid bluetooth restarts, you can disable the "input" plugin.
In order to do that, change the following line in `/lib/systemd/system/bluetooth.service
` from
`ExecStart=/usr/lib/bluetooth/bluetoothd`
to
`ExecStart=/usr/lib/bluetooth/bluetoothd --noplugin=input`
and restart Bluetooth with
`sudo systemctl daemon-reload`
`sudo systemctl restart bluetooth`
Finally we have to comment out the bluetooth restart code in the create_hid_server function in server.py.

mart1nro, 2020-03-30:
Note: Due to the restart workaround, you can not emulate multiple controllers if the bluez "input" plugin is enabled (Emulation of multiple controllers also require multiple Bluetooth adapters).

andyvandy, 2020-04-20:
I ran into issues here as well with ubuntu 19.10. 

My issue was that the adapter, `'org.bluez.Adapter1'`, wasn't being found because I hadn't setup my Pi correctly. 
I was able to resolve the issue with [this answer](https://raspberrypi.stackexchange.com/a/66581)  in case any other poor souls are searching for a fix. 

Thanks for for putting together this package- I look forward to using it!

colemickens, 2020-05-03:
What is the fundamental cause of this situation? Is this a bug? Is it a bug in bluez?

mart1nro, 2020-05-03:
The ports we have to use to emulated the controllers are well known ports reserved for Bluetooth human interface devices. The input plugin handles those devices and the ports are taken if the input plugin is enabled.

After some trial and error I'm surprised that the restart hack is working that well, but it still comes with annoying issues. It's one of the reasons the package needs to run as root. I haven't found a better solution yet, other than disabling the plugin.