mart1nro/joycontrol #11: Switch cant find the emulated controller

inksong, 2020-03-27:
Due to the failure to quickly set up the environment under macos, I installed ubuntu 18.04 in the vmware fusion in my macbook macOS 10.15.3 (19D76) and installed the relevant dependencies.
run_test_controller_buttons.py appears to run successfully, with the following output:

> sudo python3 run_test_controller_buttons.py

[02:18:34] asyncio __init__::54 DEBUG - Using selector: EpollSelector
[02:18:34] joycontrol.server create_hid_server::31 INFO - Restarting bluetooth service...
[02:18:34] joycontrol.utils run_system_command::23 DEBUG - ['systemctl restart bluetooth.service' exited with 0]
[02:18:35] joycontrol.device set_name::38 INFO - setting device name to Pro Controller...
[02:18:35] joycontrol.utils run_system_command::23 DEBUG - ['hciconfig hci0 name "Pro Controller"' exited with 0]
[02:18:35] joycontrol.device set_class::34 INFO - setting device class to 0x002508...
[02:18:35] joycontrol.utils run_system_command::23 DEBUG - ['hciconfig hci0 class 0x002508' exited with 0]
[02:18:35] joycontrol.server create_hid_server::52 INFO - Advertising the Bluetooth SDP record...

Ubuntu shares Bluetooth with the macbook, but "Pro Controller" is not found on the switch. Is there anything wrong with my setup?
I look forward to running this program under macOS, and I am willing to help.
I just want to automate fishing in animal crossing new horizons.
THX.

mart1nro, 2020-03-27:
Make sure the bluetooth adapter was set up correctly. The 'hciconfig hci0 name "Pro Controller"' command in the output changes the device name. Can you check if the device name was actually changed?
Also look if the device is really advertising. Do you find it, if you look for bluetooth devices, for example on your phone?

inksong, 2020-03-27:
I used the bluetooth signal scanning software lightblue on the iphone to monitor the correct Bluetooth signal. It looks like joycontrol has not successfully created or modified the Bluetooth device name.
I think the virtual machine sharing with the host's Bluetooth device may mess up.

inksong, 2020-03-27:
Is it possible to use your code directly under macos? Though many libs were incompatible when I tried last time.

mart1nro, 2020-03-27:
The code relies on the linux bluetooth implementation bluez to set up the adapter and advertise the service record. Also a few linux system commands are used. Those need to be ported.
I'm using native python sockets for the Bluetooth communication. I don't know if they work on macOS too.

mart1nro, 2020-03-27:
Does bluetooth work generally in the VM? You could try to set the device name manually.

inksong, 2020-03-27:
The device name has to be Pro Controller?

mart1nro, 2020-03-27:
Yes

thisiscam, 2020-03-30:
I've run into the exact same issue. https://docs.vmware.com/en/VMware-Fusion/11/com.vmware.fusion.using.doc/GUID-9C42F404-F254-4E42-A5A7-2876B408FBB1.html
says 
> Only out going connections are allowed. A virtual machine may establish a connection to a remote Bluetooth device, but remote devices do not see services that virtual machines try to advertise.
> Virtual machines cannot change the name, class, or discoverability of the host's Bluetooth adapter. The host is exclusively in control over setting whether other Bluetooth devices can discover the host, and what name is used.

So it seems like VM approach would be hard (not sure about other virtualization engines).

I also spent around a day trying to create a OSX Bluetooth HID server with no luck. This is primarily due to macOS uses IOBluetooth which is entirely different from bluez system. If anyone knows how to create a HID server on Mac, that would be a great beginning step for porting this code to Mac.

inksong, 2020-03-30:
@thisiscam 
A friend of mine gave me an inspiration. He said that there are some apps that use bluetooth to emulate joycon on Android. In this case, can we use the Android emulator to connect to the switch in macOS? There may also be some related issues as well.

inksong, 2020-03-30:
@thisiscam https://play.google.com/store/apps/details?id=com.rdapps.gamepad

thisiscam, 2020-03-30:
@inksong my take is that this will also be hard. An emulator is just another virtualization engine. The current problem seems to be that macOS uses its own Bluetooth API that there isn't many existing libraries to work with. So it will be unlikely that an Android emulator will be able to get around this issue. I could be wrong tho. 

My current solution is to use a Ubuntu machine and that is working for me now.