mart1nro/joycontrol #12: Failed to Install on Raspberry Pi 4b with Ubuntu 18.04

mistwave, 2020-03-28:
```bash
ubuntu@ubuntu:~/projects/joycontrol$ sudo pip3 install .
The directory '/home/ubuntu/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/ubuntu/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Processing /home/ubuntu/projects/joycontrol
Collecting aioconsole (from joycontrol==0.12)
Collecting dbus-python (from joycontrol==0.12)
  Downloading https://files.pythonhosted.org/packages/62/7e/d4fb56a1695fa65da0c8d3071855fa5408447b913c58c01933c2f81a269a/dbus-python-1.2.16.tar.gz (576kB)
    100% |████████████████████████████████| 583kB 52kB/s
Collecting hid (from joycontrol==0.12)
  Downloading https://files.pythonhosted.org/packages/96/ba/e1923a3f7f865cd9f3c388bf6a42b4ed149ae1a00e68f71eec49ea3d3da4/hid-1.0.4.tar.gz
Installing collected packages: aioconsole, dbus-python, hid, joycontrol
  Running setup.py install for dbus-python ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-u39lz6kt/dbus-python/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-g_1o8dwz-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    creating /tmp/pip-build-u39lz6kt/dbus-python/build
    creating /tmp/pip-build-u39lz6kt/dbus-python/build/temp.linux-aarch64-3.6
    checking whether make supports nested variables... yes
    checking whether to enable maintainer-specific portions of Makefiles... no
    checking for gcc... gcc
    checking whether the C compiler works... yes
    checking for C compiler default output file name... a.out
    checking for suffix of executables...
    checking whether we are cross compiling... no
    checking for suffix of object files... o
    checking whether we are using the GNU C compiler... yes
    checking whether gcc accepts -g... yes
    checking for gcc option to accept ISO C89... none needed
    checking whether gcc understands -c and -o together... yes
    checking how to run the C preprocessor... gcc -E
    checking for grep that handles long lines and -e... /bin/grep
    checking for egrep... /bin/grep -E
    checking for ANSI C header files... yes
    checking for sys/types.h... yes
    checking for sys/stat.h... yes
    checking for stdlib.h... yes
    checking for string.h... yes
    checking for memory.h... yes
    checking for strings.h... yes
    checking for inttypes.h... yes
    checking for stdint.h... yes
    checking for unistd.h... yes
    checking minix/config.h usability... no
    checking minix/config.h presence... no
    checking for minix/config.h... no
    checking whether it is safe to define __EXTENSIONS__... yes
    checking for a BSD-compatible install... /usr/bin/install -c
    checking whether build environment is sane... yes
    checking for a thread-safe mkdir -p... /bin/mkdir -p
    checking for gawk... gawk
    checking whether make sets $(MAKE)... yes
    checking whether make supports the include directive... yes (GNU style)
    checking whether UID '0' is supported by ustar format... yes
    checking whether GID '0' is supported by ustar format... yes
    checking how to create a ustar tar archive... gnutar
    checking dependency style of gcc... gcc3
    checking build system type... aarch64-unknown-linux-gnu
    checking host system type... aarch64-unknown-linux-gnu
    checking for native Windows host... no
    checking how to print strings... printf
    checking for a sed that does not truncate output... /bin/sed
    checking for fgrep... /bin/grep -F
    checking for ld used by gcc... /usr/bin/ld
    checking if the linker (/usr/bin/ld) is GNU ld... yes
    checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
    checking the name lister (/usr/bin/nm -B) interface... BSD nm
    checking whether ln -s works... yes
    checking the maximum length of command line arguments... 1572864
    checking how to convert aarch64-unknown-linux-gnu file names to aarch64-unknown-linux-gnu format... func_convert_file_noop
[... 69 lines cut ...]
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.6/distutils/command/install.py", line 589, in run
        self.run_command('build')
      File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-u39lz6kt/dbus-python/setup.py", line 64, in run
        cwd=builddir)
      File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/tmp/pip-build-u39lz6kt/dbus-python/configure', '--disable-maintainer-mode', 'PYTHON=/usr/bin/python3', '--prefix=/tmp/pip-build-u39lz6kt/dbus-python/build/temp.linux-aarch64-3.6/prefix']' returned non-zero exit status 1.

    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-u39lz6kt/dbus-python/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-g_1o8dwz-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-u39lz6kt/dbus-python/

```

mistwave, 2020-03-28:
After installing these 2 libs, problem solved:
```bash
sudo apt-get install libdbus-1-dev libdbus-glib-1-dev
```

mistwave, 2020-03-28:
```
ubuntu@ubuntu:~/projects/joycontrol$ sudo python3 run_test_controller_buttons.py
[07:42:23] asyncio __init__::54 DEBUG - Using selector: EpollSelector
[07:42:23] joycontrol.server create_hid_server::31 INFO - Restarting bluetooth service...
[07:42:23] joycontrol.utils run_system_command::23 DEBUG - ['systemctl restart bluetooth.service' exited with 0]
Traceback (most recent call last):
  File "run_test_controller_buttons.py", line 134, in <module>
    loop.run_until_complete(_main(controller, capture_file=capture_file, spi_flash=spi_flash))
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "run_test_controller_buttons.py", line 80, in _main
    transport, protocol = await create_hid_server(factory, 17, 19, capture_file=capture_file)
  File "/home/ubuntu/projects/joycontrol/joycontrol/server.py", line 47, in create_hid_server
    hid.powered(True)
  File "/home/ubuntu/projects/joycontrol/joycontrol/device.py", line 25, in powered
    self.properties.Set(self.adapter.dbus_interface, 'Powered', boolean)
  File "/usr/local/lib/python3.6/dist-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/local/lib/python3.6/dist-packages/dbus/proxies.py", line 147, in __call__
    **keywords)
  File "/usr/local/lib/python3.6/dist-packages/dbus/connection.py", line 653, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "Set" with signature "ssb" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

mart1nro, 2020-03-28:
Did you install dbus-python with `sudo apt install python3-dbus` before installing the package?
Maybe dbus isn't setup properly... Have you tried to restart?

mistwave, 2020-03-28:
Thank you for helping. I did install dbus-python with `sudo` and reboot the device. The exception is still existing.

mart1nro, 2020-03-28:
I haven't tried using Ubuntu on the Raspberry Pi yet. Is Bluetooth working?

mistwave, 2020-03-28:
It must be some Ubuntu's issues. It's working on Raspbian now. Thank you again!

closed 2020-06-11 by mart1nro