mart1nro/joycontrol #13: Log file format?

spacemeowx2, 2020-03-31:
I'm trying to implement NFC and meet trouble. I noticed that there is a `--log` option but I can't open it by Wireshark. Could you tell what tool can see the log file? Or is there a better way to see the log?

mart1nro, 2020-03-31:
The log file format is very basic at the moment. Here is some code to parse it:
```python
import argparse
import struct

from joycontrol.report import InputReport, OutputReport

def eof_read(file, size):
    """
    Raises EOFError if end of file is reached.
    """
    data = file.read(size)
    if not data:
        raise EOFError()
    return data

if __name__ == '__main__':
    parser = argparse.ArgumentParser()
    parser.add_argument('capture_file')
    args = parser.parse_args()

    # list of time, report tuples
    input_reports = []
    output_reports = []

    with open(args.capture_file, 'rb') as capture:
        try:
            start_time = None
            while True:
                # parse capture time
                time = struct.unpack('d', eof_read(capture, 8))[0]
                if start_time is None:
                    start_time = time

                # parse data size
                size = struct.unpack('i', eof_read(capture, 4))[0]
                # parse data
                data = list(eof_read(capture, size))

                if data[0] == 0xA1:
                    input_report = InputReport(data)
                    # normalise time
                    input_reports.append((time - start_time, input_report))
                elif data[0] == 0xA2:
                    output_report = OutputReport(data)
                    # normalise time
                    output_reports.append((time - start_time, output_report))
                else:
                    raise ValueError('unexpected data')
        except EOFError:
            pass

    print('Input reports:', len(input_reports))
    print('Output reports:', len(output_reports))

```

mart1nro, 2020-03-31:
You could also use Wireshark to capture the bluetooth communication.

spacemeowx2, 2020-03-31:
Thanks for quick response! It's very helpful to me.

closed 2020-03-31 by spacemeowx2