mart1nro/joycontrol #19: Why is root needed?

JuanPotato, 2020-04-11:
I was looking through the instructions to get this up and running and wanted to know why root was required for the script to run. Thanks for making such a cool program.

mart1nro, 2020-04-11:
- We are changing bluetooth adapter parameters (device name, device class, advertised uuids etc.) via dbus / system commands. I haven't checked if it is possible to update permissions of a user to freely change those values, but by default this requires root access.
- If you aren't disabling the Bluetooth "input" plugin, a restart of the Bluetooth stack is required (implemented using a systemctl call).

I understand running the program as root is not ideal. Maybe we can have a minimal setup script someday to avoid this, but I haven't looked into it yet.

JuanPotato, 2020-04-11:
Thank you for the quick response! I managed to get the script running on my laptop, but it took some extra work. I have to manually change the device class through hciconfig while joycontrol was running. I'll open a new issue explaining what happened. This issue has served its purpose I guess.

closed 2020-04-11 by JuanPotato