mart1nro/joycontrol #20: Bluetooth class isn't changed properly

JuanPotato, 2020-04-11:
At first joycontrol wouldnt work on my laptop (running an xps 13 9370) so I tried to figure out what was wrong. My phone showed a "pro controller" but with a laptop icon so I checked the class with hciconfig and it was set to `0x0c010c` for Laptop. I changed it to `0x002508` for Gamepad, ran joycontrol, but joycontrol seemed to revert it back to Laptop. I ran joycontrol and changed the class which worked.

mart1nro, 2020-04-11:
Interesting... It seems like the device class is reset when registering the SDP record. The Switch still connects fine for me though.

I created a pull request, where I set the device class after registering the SDP record. Please check if that works for you.

JuanPotato, 2020-05-01:
I don't think this is needed since it was working in the end after enabling pairing

closed 2020-05-01 by JuanPotato

Ruoruo-0v0, 2020-05-15:
@JuanPotato Sorry I think I might have a similar problem:

sudo python3 ./run_controller_cli.py PRO_CONTROLLER
[13:29:29] joycontrol.device set_name::69 INFO - setting device name to Pro Controller...
[13:29:29] joycontrol.device set_class::61 INFO - setting device class to 0x002508...
[13:29:29] joycontrol.utils run_system_command::75 DEBUG - ['hciconfig hci0 class 0x002508' exited with 0]
[13:29:29] joycontrol.server create_hid_server::84 INFO - Advertising the Bluetooth SDP record...
[13:29:29] joycontrol.server create_hid_server::94 INFO - Waiting for Switch to connect... Please open the "Change Grip/Order" menu.

Is this due to the same issue? How do I change the bluetooth class while it was running? Thanks!

MasonLaney, 2020-05-20:
@Ruoruo-0v0 I had the same issue and think I just got it working. I used `sudo hciconfig hciX class 0x002508` and then ran joycontrol and it worked. Running joycontrol reset it though so I had to run the command first each time I wanted to start joycontrol.

JaredEzz, 2021-04-16:
Changing the default class to 0x002508 in /etc/bluetooth/main.conf seems to be necessary to connect after the 12.0.0 update

l0nkG7, 2021-04-28:
> Changing the default class to 0x002508 in /etc/bluetooth/main.conf seems to be necessary to connect after the 12.0.0 update

How do you do that exactly

nfp0, 2023-12-04:
I'm also affected by this problem where joycontrol does not seem to set the `0x002508` class value correctly.

PR #21 (2020-12-19): changed order of device class setting
Pull request to fix #20 

Registering the Bluetooth SDP record seems to cause the device class to be reset to 0x000c010c.
Should further investigate why that happens...

Hotfix: Changed the order of setting the device class to after registering the SDP record.