mart1nro/joycontrol #25: Switch update 10.0 broke connecting to the console

NullScope, 2020-04-14:
It seems the latest update broke something in the connection protocol, I cannot run any of the test scripts as it hangs at `root _reply_to_sub_command::221 INFO - received output report - Sub command SubCommand.REQUEST_DEVICE_INFO`

This is all of the output:
```sudo python3 run_test_controller_buttons.py
[19:24:05] asyncio __init__::53 DEBUG - Using selector: EpollSelector
[19:24:05] joycontrol.device set_name::63 INFO - setting device name to Pro Controller...
[19:24:05] joycontrol.device set_class::55 INFO - setting device class to 0x002508...
[19:24:05] joycontrol.utils run_system_command::42 DEBUG - ['hciconfig hci0 class 0x002508' exited with 0]
[19:24:05] joycontrol.server create_hid_server::82 INFO - Advertising the Bluetooth SDP record...
[19:24:05] joycontrol.server create_hid_server::92 INFO - Waiting for Switch to connect... Please open the "Change Grip/Order" menu.
[19:24:08] joycontrol.server create_hid_server::96 INFO - Accepted connection at psm 17 from ('FF:FF:FF:FF:FF:FF', 17)
[19:24:08] joycontrol.server create_hid_server::98 INFO - Accepted connection at psm 19 from ('FF:FF:FF:FF:FF:FF', 19)
[19:24:10] joycontrol.protocol connection_made::107 DEBUG - Connection established.
[19:24:10] root _reply_to_sub_command::221 INFO - received output report - Sub command SubCommand.REQUEST_DEVICE_INFO```

Removed the bluetooth address just because i'm a bit paranoid

JuanPotato, 2020-04-14:
I can confirm that even trying to use the cli script with the reconnect option hangs on `REQUEST_DEVICE_INFO`

mart1nro, 2020-04-15:
Gonna look into it. At least the relay script is still working.

mart1nro, 2020-04-15:
I just pushed a quick fix that seems to solve the issue for now. It's a hack though and not ideal. Gonna investigate further.

NullScope, 2020-04-15:
I can confirm its working now, thanks!

I also looked into your fix and I see you are sending empty inputs, I also noticed that before that the controller does pair, maybe you should send L + R input like you do for pairing a normal controller?

closed 2020-06-11 by mart1nro