mart1nro/joycontrol #27: Is there has anyway to report amiibo data?

HxBreak, 2020-04-18:
I played animal crossing recently,I am looking for some code can emulate player to complete boring things in game, I found this is the best ,After add some easy function, I am thinking how can i use amiibo data, Can you give me some hints or helps to implemente it?

**Thanks For Your Project, That 's really usefully.**

spacemeowx2, 2020-04-18:
I was working on this and only one step to complete. 

https://github.com/spacemeowx2/joycontrol/tree/nfc-r1

JuanPotato, 2020-04-18:
https://github.com/CTCaer/Nintendo_Switch_Reverse_Engineering/tree/ir-nfc
https://gist.github.com/CTCaer/69ce224521d1ee4b153b250d9cbbb6a9

You may find these links helpful, not sure if you've seen them

HxBreak, 2020-04-19:
thanks all your helps, I am ready to take a look.

mart1nro, 2020-04-20:
@spacemeowx2 That's great, I'm going to check it out. If you have a working nfc implementation feel free to pull request :)

mart1nro, 2020-04-20:
I'm reopening this issue until nfc is implemented.

HxBreak, 2020-04-20:
@spacemeowx2 May i have a chat with you, I got stuck, I sent a email to your Google Mail Box

spacemeowx2, 2020-04-21:
I just complete nfc ntag215 reading, it's a little dirty but works. https://github.com/spacemeowx2/joycontrol/tree/nfc-r2

spacemeowx2, 2020-04-21:
I'll clean the code and make a pr later

PR #29 (2020-05-04): feat: send amiibo report
Closes: #27 

change requests are welcome
closed 2020-05-04 by mart1nro