mart1nro/joycontrol #30: cannot dump the whole spi with the script.

wjm2038, 2020-04-26:
I tried to connect my left or right joy-con through Bluetooth by using the bluetoothctl's pair, trust, and connect command. However, when I try to use the script to log the spi, it never runs though all the offset and success. It will just be stuck in the middle and stop there.  Most of the time, I'll get an AssertionError.
I got a raspberry pi 3b and my os is
`OS: Raspbian 10 buster`

choss, 2020-07-21:
use: https://github.com/CTCaer/jc_toolkit on a windows box maybe?

wjm, 2020-07-25:
> use: https://github.com/CTCaer/jc_toolkit on a windows box maybe?

Thanks for the help. I already use this toolkit to dump the spi after the provided method didn't work. The spi works fine but I just wonder if this bug is fixable or there is something wrong with my setup