mart1nro/joycontrol #31: input commands lag periodically

typ127, 2020-04-27:
Hi

First thank you very much for sharing this wonderful project.

I managed to use the joycontrol library to connect a selfmade arcade controller with buttons, digital and analog sticks to the switch. It works perfect as long as i only connect one controller to the switch. When there is another controller connected to the switch (e.g. the original joycons, does not matter which ones), I realize some laggy behaviour. Commands will become delayed for some ms. It feels like data becomes queued up and processed later all at once. It happens periodically every one and a half seconds. It is best measurable in the stick calibration mode on the switch. The point slows down every period and speeds up imediatelly after. Buttons behave the same.

I am not sure where exactly to look at to solve this problem. Could it be related to asyncio or the send state frequency (120Hz)? But why does it only occur when a second controller is connected, what is different in this situation? Did someone else face this problem?

I use a Raspi 3B+ with Buster.

mart1nro, 2020-04-30:
> I managed to use the joycontrol library to connect a selfmade arcade controller with buttons, digital and analog sticks to the switch. 

That's really cool.

The delay problem you are describing is also noticeable while pairing a controller in the "Change Grip/Order" menu. I didn't know that it also happens when multiple controllers are connected. 

I don't have a solution for that yet, but I'll keep you updated.

JuanPotato, 2020-05-04:
I've been writing joycontrol in rust for fun and to see if certain issues would go away if the code was faster. Currently the program acts in relay mode and I see this issue occur when I have multiple controllers connected. My current set up is I have a left joycon directly connected to the switch via bluetooth, and the right one is relayed to my laptop and then to the switch.

When I don't connect the left joycon, my relayed right one works fine. Once the left one is connected, it lags a bit. I can disconnect the left joycon and then suddenly it works normally again.

JuanPotato, 2020-05-06:
More updates to add to this. I have my pro controller wireless connected relayed through my laptop. Switch has joycons attached. When I disconnect the joycons so that they become bluetooth connected, my laptop received an HCI event "Max slots changed". My guess says I need to forward specific hci events like this one over to the controller. Gotta look into this more. I'll put further updates on my repo to avoid spamming here 

https://github.com/JuanPotato/joycontrolrs

packdstack, 2020-09-13:
Not sure how the joycons really deal with this, but after playing around with it, I noticed it's much more responsive at the CGO (Change Grip/Order) screen when you only report input at 10hz.  It may just be my luck, but the Switch kicked it out pretty quickly at that refresh rate once I left that screen.  (Piled up rumble reports, I guess.)  At 15hz, it's less responsive, but acts a little more normal at the CGO screen than @ 60hz and didn't kick it out later, but, again, that may be luck.

Then I tried decoupling send and receive; receive @ 60hz (for rumble) and send @ 10hz.  The CGO screen seemed normal.  Never got kicked out afterwards.  It may be worth trying to report @ 10hz minimum (or not at all) when nothing's going on, then ramp it up to 60hz when there's actually input worth reporting.

Ideally, you wouldn't have to throttle anything, but here we are.

JuanPotato, 2022-05-04:
https://stackoverflow.com/questions/67185675/bluez-python-buffering-causes-laggy-bluetooth-connection

Might be a good idea to try this out. Throwing this here so I don't lose it.