mart1nro/joycontrol #33: Suggested OS and board

diesel701, 2020-05-01:
Hi, very interesting project.
I would like to create a sort of an adapter to connect the Dualshock 4 to the Switch vie Bluetooth.
My idea is to take a Raspberry Pi Zero W and connect it with USB to the Dualshock 4. Then use your program to send the commands to the Switch.

So my questions:
- Would the Raspberry Pi Zero enough powered for the task?
- If I want to test the program with my PC firstly, which OS should I use? I have read that Ubuntu has some problems.

Thanks a lot

mart1nro, 2020-05-01:
Hey, Ubuntu on pc works fine for me. The issue with Ubuntu was on the raspberry pi, but I haven't checked nor confirmed it yet (I use Raspian atm).
I don't know if the Raspberry Pi Zero is powerful enough, but it shouldn't be too much off a daunting task.

diesel701, 2020-05-01:
Thanks for now. In the next days I will try with Ubuntu and then I will order the Raspberry Zero. Would be awesome to emulate my Dualshock 4 without usb dongles on the Switch.

diesel701, 2020-05-02:
Hi, I'm trying it on Ubuntu 20.04 and I have this error when I run the python script:
`Traceback (most recent call last):
  File "run_controller_cli.py", line 11, in <module>
    from joycontrol import logging_default as log, utils
  File "/home/ubuntu/Desktop/joycontrol-master/joycontrol/utils.py", line 5, in <module>
    import hid
  File "/usr/local/lib/python3.8/dist-packages/hid/__init__.py", line 30, in <module>
    raise ImportError(error)
ImportError: Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 libhidapi.dylib hidapi.dll libhidapi-0.dll
`
How can I fix it? Thanks

mart1nro, 2020-05-03:
Hi, sorry you also had to install install libhidapi-hidraw0. It is probably part of the libhidapi-dev package you mentioned.

@TylerCarberry kindly made a pull request and it is now added in the Readme.

diesel701, 2020-05-03:
Great.
I have created a script for my PS3 controller and it works great (I have discovered my PS4 controller has a left stick drift, so I have to fix that before using it with joycontrol).

Just to know, I have created a new function "button_release", so now I can keep pressed my button and scroll.
Here the code:
 
```
async def button_push(controller_state, *buttons):
    if not buttons:
        raise ValueError('No Buttons were given.')

    button_state = controller_state.button_state

    for button in buttons:
        # push button
        button_state.set_button(button)

    # send report
    await controller_state.send()

async def button_release(controller_state, *buttons):
    if not buttons:
        raise ValueError('No Buttons were given.')

    button_state = controller_state.button_state

    for button in buttons:
        # release button
        button_state.set_button(button, pushed=False)

    # send report
    await controller_state.send()

```

Now I have to work on "sticks"

mart1nro, 2020-05-04:
> I have created a script for my PS3 controller and it works great

@diesel701, that's great. I was planing on implementing xbox controller support soon. Are you willing to share your progress?

diesel701, 2020-05-06:
Yes, I have implemented the joystick too now.
When I am time I will fork and upload the new script

diesel701, 2020-05-07:
I have uploaded my script and my little modification of controller_state.py in a branch of the forked repo (sorry I am new to git).
[HERE](github.com/diesel701/joycontrol/blob/controllers/use_controller.py) there's the script I wrote to use PS3 or PS4 controller as an input for Nintendo Swtich.

I tested it in USB and Bluetooth mode and it works great! 

Next week I will work on Xbox360 controller

carrotcn, 2020-05-09:
FYI - similar to the unstability on RPi 4, I encountered issue when running on a RPi Zero W as well. The connection drops after 10-50 minutes. For the sake of simulating Amiibo that is not a big deal but could be a problem if one is trying to simulate a list of actions and repeat for hours..

diesel701, 2020-05-09:
> FYI - similar to the unstability on RPi 4, I encountered issue when running on a RPi Zero W as well. The connection drops after 10-50 minutes. For the sake of simulating Amiibo that is not a big deal but could be a problem if one is trying to simulate a list of actions and repeat for hours..

Ubuntu or Rasbian
It seems an UART bug in kernel that is fixed in one of the latest releases. 
Try this:
```
$ sudo apt-get update
$ sudo apt-get install raspberrypi-kernel
$ sudo reboot

carrotcn, 2020-05-09:
> Ubuntu or Rasbian
> It seems an UART bug in kernel that is fixed in one of the latest releases.
> Try this:
>
> ```
> $ sudo apt-get update
> $ sudo apt-get install raspberrypi-kernel
> $ sudo reboot
> ```

Rasbian. I tried updating the kernel as suggested but the issue persists.. I kept simulating the action of pressing A button every 3 seconds before the connection failure occurred..

packdstack, 2020-06-27:
I had issues with wifi and bluetooth competing on RPi Zero W.  Bluetooth lagged occasionally until I disabled wifi.  (They share the same chip.)  I ended up making my app auto-disable wifi when it was go-time, then re-enable it when the system wasn't ready.  Not sure if that's still an issue; haven't really looked at it in a while.

ebisbe, 2020-08-04:
> FYI - similar to the unstability on RPi 4, I encountered issue when running on a RPi Zero W as well. The connection drops after 10-50 minutes. For the sake of simulating Amiibo that is not a big deal but could be a problem if one is trying to simulate a list of actions and repeat for hours..

That's the issue I have. I got random disconnects from a Raspberry 4