mart1nro/joycontrol #36: Run controller rest api?

benhid, 2020-05-04:
Hello,

I have successfully tested `joycontrol` on a Raspberry Pi Zero W (I like its portability and the fact that it has BT and WiFi connectivity built-ins) but every time I want to run the client I have to turn on my computer, create a new _ssh_ session, run the script, etc . It is easy for most of us, but not for some people with limited understanding of Linux (in general).

How difficult could it be to implement a REST API with very basic functionality (just some endpoints to push buttons and check the state)?

Thank you.

diesel701, 2020-05-07:
I would like to use a Pi Zero W too. How are the performances? Is there some lag?

Anyway, I think you can automate the script starting it when the system boot. Then use a while loop for the connection with the Switch, or maybe use some sort of button input to enable the sync. 
Obviously you need to write your own script. 
I am working on it during my free time, but I'm waiting for my Pi Zero W for now.

cool00geek, 2020-05-29:
I am currently working on a web app to control the Switch from a web browser, and in the process I have exposed a basic API that pushes buttons and sets the joystick positions. At the moment I do not have any way to check the state though. If you have any feedback it would be great!

https://github.com/cool00geek/fakecon

I do believe, like @diesel701 said, to add extra functionality separately instead of baking it into joycontrol directly.

choss, 2020-07-23:
I have written one which resides over here: https://github.com/choss/joycontrol_rest_api
You are happy to test it and leave feedback.

dentedghost, 2020-07-26:
@choss please check with @mart1nro to see if we can have it added to this repo to extend usability.

choss, 2020-07-27:
@dentedghost I think @mart1nro is right here! So if he would like a PR he can just comment right here :) 
Also I'm taking suggestions and pull-requests, too, if someone wants to add documentation or a setup script to get it easier usable. (Like a bash script with a one file setup)

dentedghost, 2020-08-01:
@choss going to test it out.  I want to convert https://github.com/SkyoKen/RasCon_NS to leverage your API interface.

moeiscool, 2020-10-15:
for anyone install interested in this i started making this for my niece to play my nintendo while shes at her house https://gitlab.com/moeiscool/web-nintendo-switch-controller

so far ive gotten the latency down to 0.5 seconds but its still not good enough.

I am testing on Jetson Nano with hardware accelerated FFmpeg (decoding only).

mixaal, 2020-11-29:
Hi, in case you're interested here's a forked version which can use raspberry pi controller to handle nintendo switch: https://github.com/mixaal/joycontrol 

Internally it uses rest api to control NS from multiple input sources.

juharris, 2020-12-19:
Wow, looks like a few of us have implemented the same thing. I have a web socket API for joycontrol, and a public site set up where you enter the URL for your sever hosting joycontrol. You can even record macros!
Full details and example videos here: https://github.com/juharris/switch-remoteplay