mart1nro/joycontrol #38: Amiibo does not work

carrotcn, 2020-05-07:
Hello,

I set up the project on RPi Zero W and the button controls work perfectly. However when I tried to load the amiibo dump file (.bin) with the cmd> amiibo <filepath> command the Switch does not seem to sense anything. There is no error thrown on the console output either. Is there any specific format that the amiibo dump file shall follow, or there is any special config that I am not aware of..?

Thanks!

carrotcn, 2020-05-07:
An update from my side - 

I attempted multiple combinations of events and finally noticed that if I send multiple unrelated actions like pushing A, B buttons after loading the amiibo .bin file, the amiibo will eventually be recognized by the switch. However if I just load the amiibo .bin file without doing any subsequent button event push it will not be recognized alone...

Thanks.

JuanPotato, 2020-05-07:
I believe this is because if you weren't doing anything with the fake controller, the switch doesn't know to read from it. So it reads from the joycons or whatever controllers you have attached. If you start using the fake controller, it assumes that it is your primary controller and you want to use it for amiibo. You can test this out more in the system settings amiibo section. If the fake controller is the only one that is connected, then it should immediately get the amiibo data

carrotcn, 2020-05-07:
Yep that makes sense. Thank you!

closed 2020-05-07 by carrotcn