mart1nro/joycontrol #40: Syntax Error?

sorayori, 2020-05-07:
Whenever I run the "sudo python3 run_controller_cli.py PRO_CONTROLLER" command, it gives me this syntax error:

Traceback (most recent call last):
  File "run_controller_cli.py", line 10, in <module>
    from joycontrol import logging_default as log, utils
  File "/home/pi/Downloads/joycontrol-master/joycontrol/logging_default.py", line 31
    name_of_file = today.strftime(f'%Y-%m-%d_%H-%M_{logfile_name}.log')
                                                                     ^
SyntaxError: invalid syntax

i'm probably just really dumb, but what can i do to fix this?

JuanPotato, 2020-05-07:
Can you run `python3 --version` and paste the output here?

You're not dumb, for some reason the script isn't being read correctly.

sorayori, 2020-05-07:
> Can you run `python3 --version` and paste the output here?
> 
> You're not dumb, for some reason the script isn't being read correctly.
I'm currently running Python 3.5.3

JuanPotato, 2020-05-07:
That explains it, f-strings were introduced in python 3.6.
The line it breaks at contains `f'%Y-%m-%d_%H-%M_{logfile_name}.log'` which is an f-string that makes formatting variables into strings easy in python. 3.5.3 is a really old version. If you can update, I recommend doing that, otherwise I can submit a pull request that turns these f-strings into older format function calls so that it works with older python version.

sorayori, 2020-05-07:
Thank you for the help, i'll see if i can update now!

closed 2020-05-07 by sorayori