mart1nro/joycontrol #43: Can't register Amiibo

smthless, 2020-05-09:
Hello!
First of all, thank you very much, Joycontrol works perfectly in games where you don't have to register amiibo to use it.
However, I can't make it work whenever registration is involved. Switch recognizes amiibo but when it comes to registration I get "registration failed" pop-up on switch.
I'm running Ubuntu 20.4 on my laptop with integrated bluetooth adapter. 

Could you please kindly advise me if anything can be done to resolve this?

Best wishes,
Sergio

mart1nro, 2020-05-09:
Hi, registration is not supported atm.

yaolet, 2020-05-09:
amazing project!  any timeline for amiibo support?

carrotcn, 2020-05-11:
> amazing project! any timeline for amiibo support?

Actually you can use Amiibo with this project, to summon chests in BotW and visitors in ACNH, as long as you you are not trying to register it to link it to specific users in system menu...

NNThomasL, 2020-05-11:
> > amazing project! any timeline for amiibo support?
> 
> Actually you can use Amiibo with this project, to summon chests in BotW and visitors in ACNH, as long as you you are not trying to register it to link it to specific users in system menu...

I have not been able to get Amiibos to work in BotW. When I connect the virtual controller as a pro controller or right joycon, it disconnects as soon as I get into the game. If I am already in the game and use the -r argument, it disconnects as soon as I leave the menu where you choose which controller to play with.

Tenkoni, 2020-05-11:
> I have not been able to get Amiibos to work in BotW. When I connect the virtual controller as a pro controller or right joycon, it disconnects as soon as I get into the game. If I am already in the [...]

Seems like in BotW is a bit buggy, but I have been successful a it. You need to let everything ready for the scan (don't use the sheikah amiibo module yet), run joycontrol and the switch will pop out a connection screen, try sending button presses in the joycontrol console, it will take a bit to detect the controller in the switch. After that just use the virtual controller, not the physical joycons or it will disconnect the virtual one. After that just load the amiibo and press L (using the virtual procontroller).
And that should do it.