mart1nro/joycontrol #46: Docker Image / 2min setup :raised_hands:

Pierozi, 2020-05-14:
If some of you don't want loose time to find right python version and so on.

1. Need docker installed
2. Need Bluetooth device ( mine is USB key )
3. Clone this repository ( mart1nro/joycontrol ) and move your terminal on this root folder
4. Run the docker image
```
docker run -ti --privileged --net=host --volume $(pwd):/joycontrol pierozi57/python-ble:latest
```
5. Start DBUS
```
service dbus start
```
6. Start Bluetoothd without input (https://github.com/mart1nro/joycontrol/issues/8)
```
bluetoothd --noplugin=input > /var/log/bluetoothd.log 2>&1 &disown
```
7. Launch  Joycontrol
```
python ./run_controller_cli.py PRO_CONTROLLER
```
8. Do whatever you have to do

lilithGao, 2020-05-20:
If I use docker to run on mac, I will get an error: Address family not supported by protocol，
What device are you running docker on?

Pierozi, 2020-05-25:
It must be because you already have `bluetoothd` running on your OSX, and it will be tricky to kill as it restarts automatically.

acidhax, 2020-05-26:
I have the same issue on Mac, I have no time to research it, but it seems incompatible

pronoiac, 2020-07-16:
You might be able to stop bluetoothd on Mac with: 

`sudo launchctl stop com.apple.bluetoothd`

FedericoHeichou, 2022-04-24:
Dockerfile
```
FROM python:3.9

RUN apt update \
    && apt install -y libglib2.0-dev libhidapi-hidraw0 libhidapi-libusb0 libdbus-1-dev bluetooth usbutils git

RUN pip3 install hid aioconsole crc8 dbus-python
RUN cd / && git clone https://github.com/mart1nro/joycontrol.git

COPY docker-entrypoint.sh /usr/local/bin/docker-entrypoint.sh
RUN chmod +x /usr/local/bin/docker-entrypoint.sh

WORKDIR /joycontrol
ENTRYPOINT ["/usr/local/bin/docker-entrypoint.sh"]
CMD ["python3", "./run_controller_cli.py", "PRO_CONTROLLER"]
```

docker-entrypoint.sh
```
#!/bin/bash
service dbus start
echo "Starting bluetoothd"
bluetoothd --noplugin=input > /var/log/bluetoothd.log 2>&1 &
echo "Bluetoothd started"

exec "$@"
```

```
docker build --tag=joycontrol .
docker run -it --privileged --net=host --name=joycontrol -v $(pwd)/amiibo:/amiibo joycontrol
```

You may need to change the `git clone` url