mart1nro/joycontrol #50: Rumble support

yaolet, 2020-05-17:
Right now in the console I don't see any rumble commands being printed.  any support right now or planning?

mart1nro, 2020-05-17:
No support yet, but would be nice to have.

dentedghost, 2020-07-23:
@mart1nro Can you point me in the direction to capture Rumble?  I'd be willing to give it a shot to add it.

mart1nro, 2020-07-23:
@dentedghost  Sure! That would be great.

There is some information on  [dekuNukem/Nintendo_Switch_Reverse_Engineering ](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering)

The rumble data bytes in an output report (what the Switch sends) are described here:
https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/bluetooth_hid_notes.md#rumble-data

Here is some information and code, on how those bytes are encoded:
[rumble_data_table.md](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/rumble_data_table.md)
For the joycontrol library we would need to do this in reverse.

There is also byte 12 in the [Standard input report format](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/bluetooth_hid_notes.md#standard-input-report-format), send by the controller, which is described as Vibrator input report byte, but I don't really understand what it does.

I also just pushed a branch (https://github.com/mart1nro/joycontrol/tree/rumble_experiments) where I started experimenting with rumble a while back.
There, I basically tried to implement the algorithms from the [rumble_data_table.md](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/rumble_data_table.md) in the `set_left_rumble_data` and `set_right_rumble_data` functions of the `OutputReport` class (report.py).
I don't think the implementation is fully correct, but I added a script to send some vibration reports to the right joycon (`scripts/joycon_rumble.py`) and it lets the controller vibrate.

dentedghost, 2020-07-27:
Research update:
@littlefisher619:  Was able to parse the output report to detect Rumbles in  https://github.com/Littlefisher619/MyCTFChallenges/blob/master/CTFChallengeSwitch/Challenge2/writeup/parse_capture.py

Questions: With # elif output_report_id == OutputReportID.RUMBLE_ONLY: will this show when any Rumble happened?

As an MVP I think just reporting if any rumble occurred,  would allow most who are wait for an Rumble then to send the appropriate commands following that Rumble.   Thoughts on best way to just provide a flag if Rumble or Not that could be checked?

pvanderlinden, 2020-12-13:
Feedback on my WIP PR would be good, it sort of implements rumble decoding