mart1nro/joycontrol #51: ModuleNotFoundError: No module named 'aioconsole'

cb2T5rju, 2020-05-18:
I previously had no issues emulating a controller on Ubuntu 19.10, but I have subsequently upgraded to Ubuntu 20.04 LTS. When I run `sudo python3 ./run_controller_cli.py PRO_CONTROLLER` I am now getting the following output:
`Traceback (most recent call last):
  File "./run_controller_cli.py", line 8, in <module>
    from aioconsole import ainput
ModuleNotFoundError: No module named 'aioconsole'`

mart1nro, 2020-05-19:
Maybe some python packages got deleted during the Ubuntu upgrade.
Try installing the joycontrol package again by running
`sudo pip3 install --upgrade .`
in the joycontrol folder

legoironman1234, 2020-05-20:
what do you mean when you say install the package in the folder

hammy3502, 2020-05-20:
I remembered encountering a similar problem, I think it's a missing dependency from `setup.py`?

Either way, you should be able to solve it with `pip3 install aioconsole`

closed 2020-06-11 by mart1nro

00-Python, 2023-05-18:
any update on this issue? i am having the same problem but i have aioconsole installed

```
(joycontrol) zero@zero-Lappy:~/Desktop/joycontrol$ sudo python3 run_controller_cli.py PRO_CONTROLLER
Traceback (most recent call last):
  File "/home/zero/Desktop/joycontrol/run_controller_cli.py", line 8, in <module>
    from aioconsole import ainput
ModuleNotFoundError: No module named 'aioconsole'

(joycontrol) zero@zero-Lappy:~/Desktop/joycontrol$ pip freeze
aioconsole==0.6.1
dbus-python==1.3.2
hid==1.0.5
joycontrol==0.15
(joycontrol) zero@zero-Lappy:~/Desktop/joycontrol$ 

```

nick16, 2023-05-18:
@00-Python 
I solved my missing dependency issues by installing them within a virtual python environment, and running the script within the virtual environment.

00-Python, 2023-05-18:
That is what i am doing and it doesn't seem to work, when i run `pip freeze` or `pip3 freeze` it says its installed but when i run `python3 -m pip freeze` it returns the packages i have installed on my system python not the venv.  so i tried installing it with  with `python3 -m pip install aioconsole` and i get the following error:
```
(joycontrol) zero@zero-Lappy:~/Desktop/joycontrol$ sudo python3 -m pip install aioconsole
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

So i try to do as it says but python3-aioconsole is not a package according to apt.

```
(joycontrol) zero@zero-Lappy:~/Desktop/joycontrol$ sudo apt install python-aioconsole
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python-aioconsole
(joycontrol) zero@zero-Lappy:~/Desktop/joycontrol$ sudo apt install python3-aioconsole
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3-aioconsole
```

nick16, 2023-05-26:
remove the package outside the venv then reinstall inside the venv ¯\_(ツ)_/¯