mart1nro/joycontrol #53: OSError: [Errno 97] Address family not supported by protocol

lgb1234a, 2020-05-21:
Error reported when i ran `sudo python3 ./run_controller_cli.py PRO_CONTROLLER`
The whole console log:
```
Traceback (most recent call last):
  File "./run_controller_cli.py", line 267, in <module>
    loop.run_until_complete(
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "./run_controller_cli.py", line 182, in _main
    transport, protocol = await create_hid_server(factory, reconnect_bt_addr=args.reconnect_bt_addr,
  File "/home/chenyn/joycontrol/joycontrol/server.py", line 45, in create_hid_server
    ctl_sock = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_SEQPACKET, socket.BTPROTO_L2CAP)
  File "/usr/lib/python3.8/socket.py", line 231, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 97] Address family not supported by protocol
```

My operate system info:
        **Windows 10 professional edit installed by BootCamp on Macbook pro 15'** 
        **Unbunto 20.04 LTS installed from Microsoft store**
Is there any problem occur with Mac OS to do this? 
Thx for any advice.

yaolet, 2020-05-25:
i guess thats something to do with the bluez dependency which is absent in win10?

lgb1234a, 2020-05-25:
> i guess thats something to do with the bluez dependency which is absent in win10?

You mean the win10 installed on Mac doesn't respond to the **bluez** protocol?

yaolet, 2020-05-30:
bluez is a bluetooth service under linux which this project is based on.  So python in windows won't work.

jpas, 2020-06-02:
To clarify a bit further, having Ubuntu installed from the Microsoft Store does not have the capability to support bluez as it relies on a kernel module that directly accesses the bluetooth controller.
Ubuntu running under Windows Subsystem for Linux does not have direct access to the bluetooth controller.

The fact that you are using a Mac does not make a difference here.

Raymondma-public, 2022-05-15:
Seems like new WSL linux kernel configuration allows Linux on WSL to use bluetooth with a 3rd-party library usbipd-win.
But I am whether it could help on this issue.

Video:
https://www.youtube.com/watch?v=I2jOuLU4o8E&ab_channel=MicrosoftDeveloper

Library usbipd-win
https://github.com/dorssel/usbipd-win