mart1nro/joycontrol #54: Does not support pairing with another Joycon?

suyeonahn, 2020-05-26:
Although this code can emulate a left/right joycon, it will not pair with another physical joycon. I tried using my left joycon and emulating the right joycon, but the switch will connect to the emulated joycon in 2p mode (sl/sr), and the left joycon is assigned as a second player. I was wondering if there is a way to modify the code such that an emulated joycon can be paired with an existing one?

mart1nro, 2020-05-27:
You would need to change line 442 in [joycontrol/protocol.py](https://github.com/mart1nro/joycontrol/blob/721646a7ec10231490bcf788b3bdde10d8c2007f/joycontrol/protocol.py#L443)

During pairing the Switch asks how long the trigger buttons where pressed and decides the joycon layout accordingly. 

We should probably implement a flag somewhere.

suyeonahn, 2020-05-28:
Thank you very much! I was able to add a quick if statement for JOYCON_R mode and the appropriate button trigger.

`if self.controller == Controller.JOYCON_R:
                input_report.sub_0x04_trigger_buttons_elapsed_time(R_ms=3000)`

 My physical and emulated joycons paired successfully! It was a bit tricky trying to time my physical press with the emulated press, but I found that just spamming the L button eventually gets the job done.

closed 2020-05-28 by suyeonahn