mart1nro/joycontrol #65: broken on 10.0.4

yaolet, 2020-07-07:
with 10.0.4 firmware, the emulation will be disconnected on leaving the Change grip/order screen.

tested with rpi & x86 ubuntu 20.04

mart1nro, 2020-07-07:
I tested it with 10.0.4 on my Laptop and it It still works.

I'm going to check it on my Raspberry Pi 4 as well once I have time and report back.

dpxcc, 2020-07-07:
duplicate to https://github.com/mart1nro/joycontrol/issues/26?

yaolet, 2020-07-07:
> duplicate to #26?

yeah seems like the same issue.  Its very odd that the same pi worked last week with the same code. Suddenly it just stopped working

yaolet, 2020-07-19:
tested again on 10.1.0 on x86 ubuntu 20.04

problem still exists

Poohl, 2021-03-15:
When did this problem start to appear? I'm suddenly having the same problem on ubuntu 20.04 since 2 weeks ago (using a laptop). The odd thing is the connection only drops after you send the first input, until then it's totally fine (but useless).

Ubuntu version: 20.04
Switch version: 11.0.1