mart1nro/joycontrol #66: Connection Lost?

DeathUriel, 2020-07-09:
First time running this project. After some tries I made it work. (too stupid to just read the damn readme first)
But then when I got to the message "Waiting for Switch to connect..." it doesn't seem to work. Like I have two switches here, I tried both and nothing shows on them. The code says it connected to something it seems but doesn't and then crashes. I tried both on Debian (AMD64) and a Raspberry Pi 4 running Ubuntu Server (ARM64) and nothing seems to be different.
Also, should mention that I did see the "Pro Controller" with an notebook icon inside my android's bluetooth pairing screen. Tried to connect to it and nothing happened. But all tests after that were with my android's bluetooth turned off just to be safe.

Here is my stack trace:

[20:53:49] joycontrol.server create_hid_server::94 INFO - Waiting for Switch to connect... Please open the "Change Grip/Order" menu.
[20:53:57] joycontrol.server create_hid_server::98 INFO - Accepted connection at psm 17 from ('DC:68:EB:6E:A5:40', 17)
[20:53:57] joycontrol.server create_hid_server::100 INFO - Accepted connection at psm 19 from ('DC:68:EB:6E:A5:40', 19)
[20:53:57] joycontrol.protocol connection_made::112 DEBUG - Connection established.
[20:53:57] joycontrol.transport read::105 ERROR - No data received.
[20:53:57] joycontrol.protocol connection_lost::117 ERROR - Connection lost.
[20:53:57] asyncio default_exception_handler::1707 ERROR - Task exception was never retrieved
future: <Task finished name='Task-12' coro=<L2CAP_Transport.close() done, defined at /home/ubuntu/samba/joycontrol/joycontrol/transport.py:172> exception=AttributeError("'NoneType' object has no attribute 'cancel'")>
Traceback (most recent call last):
  File "/home/ubuntu/samba/joycontrol/joycontrol/transport.py", line 179, in close
    if self._read_thread.cancel():
AttributeError: 'NoneType' object has no attribute 'cancel'
[20:53:58] joycontrol.transport write::155 ERROR - [Errno 107] Transport endpoint is not connected
^CTraceback (most recent call last):
  File "run_controller_cli.py", line 277, in <module>
    loop.run_until_complete(
  File "/usr/lib/python3.8/asyncio/base_events.py", line 603, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 570, in run_forever
    self._run_once()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 1823, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.8/selectors.py", line 468, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
[20:54:24] asyncio default_exception_handler::1707 ERROR - Task exception was never retrieved
future: <Task finished name='Task-11' coro=<_send_empty_input_reports() done, defined at /home/ubuntu/samba/joycontrol/joycontrol/server.py:17> exception=NotConnectedError(OSError(107, 'Transport endpoint is not connected'))>
Traceback (most recent call last):
  File "/home/ubuntu/samba/joycontrol/joycontrol/transport.py", line 153, in write
    await self._loop.sock_sendall(self._itr_sock, _bytes)
  File "/usr/lib/python3.8/asyncio/selector_events.py", line 439, in sock_sendall
    n = sock.send(data)
OSError: [Errno 107] Transport endpoint is not connected

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/samba/joycontrol/joycontrol/server.py", line 20, in _send_empty_input_reports
    await transport.write(report)
  File "/home/ubuntu/samba/joycontrol/joycontrol/transport.py", line 157, in write
    raise NotConnectedError(err)
joycontrol.transport.NotConnectedError: [Errno 107] Transport endpoint is not connected

someonewithpc, 2020-07-23:
What firmware are you running?

DeathUriel, 2020-08-04:
It was on latest at the day of writting. Is there any cheap bluetooth dongle confirmed to work?

ebisbe, 2020-08-06:
> It was on latest at the day of writting. Is there any cheap bluetooth dongle confirmed to work?

Did you find any? From what I've read there's some issues with the bluetooth implementation in the raspbian OS as you can read here https://github.com/raspberrypi/linux/issues/2832

DeathUriel, 2020-08-06:
> > It was on latest at the day of writting. Is there any cheap bluetooth dongle confirmed to work?
> 
> Did you find any? From what I've read there's some issues with the bluetooth implementation in the raspbian OS as you can read here [raspberrypi/linux#2832](https://github.com/raspberrypi/linux/issues/2832)

I don't think that applies.
First of all I believe we were talking about switch's firmware. At least that's what I answered.
Also I don't use Raspberry OS, I tried both on debain (desktop x64) and Ubuntu Server (raspberry Pi). So an OS problem like that reported would only make sense if it is something that happens in all debian distributions in any architecture.