mart1nro/joycontrol #75: Transport endpoint is not connected

ebisbe, 2020-08-04:
Randomly disconnects and when I try to connect again with a Raspberry 4 the result is this:
I'm using github.com/choss/joycontrol_rest_api/
```
2020-08-04 10:34:56,761 - INFO - 192.168.0.104:55464 - "POST /controller/connect HTTP/1.1" 200
INFO:root:received output report - Sub command SubCommand.SET_SHIPMENT_STATE
INFO:root:received output report - Sub command SubCommand.SPI_FLASH_READ
INFO:root:received output report - Sub command SubCommand.SPI_FLASH_READ
INFO:root:received output report - Sub command SubCommand.SET_INPUT_REPORT_MODE
INFO:joycontrol.protocol:Setting input report mode to 0x30...
INFO:root:received output report - Sub command SubCommand.TRIGGER_BUTTONS_ELAPSED_TIME
INFO:root:received output report - Sub command SubCommand.SPI_FLASH_READ
INFO:root:received output report - Sub command SubCommand.SPI_FLASH_READ
INFO:root:received output report - Sub command SubCommand.SPI_FLASH_READ
INFO:root:received output report - Sub command SubCommand.SPI_FLASH_READ
INFO:root:received output report - Sub command SubCommand.SPI_FLASH_READ
ERROR:asyncio:Exception in callback <function create_error_check_callback.<locals>.callback at 0xb4e34588>
handle: <Handle create_error_check_callback.<locals>.callback>
Traceback (most recent call last):
  File "uvloop/cbhandles.pyx", line 70, in uvloop.loop.Handle._run
  File "/usr/local/lib/python3.7/dist-packages/joycontrol/utils.py", line 58, in callback
    future.result()
  File "/usr/local/lib/python3.7/dist-packages/joycontrol/protocol.py", line 155, in input_report_mode_full
    data = await reader
  File "/usr/local/lib/python3.7/dist-packages/joycontrol/transport.py", line 99, in read
    data = await self._loop.sock_recv(self._itr_sock, self._read_buffer_size)
  File "uvloop/loop.pyx", line 2430, in sock_recv
  File "uvloop/loop.pyx", line 926, in uvloop.loop.Loop._sock_recv
TimeoutError: [Errno 110] Connection timed out
ERROR:joycontrol.transport:[Errno 107] Transport endpoint is not connected
ERROR:joycontrol.protocol:Connection lost.
ERROR:asyncio:Task exception was never retrieved
future: <Task finished coro=<_send_empty_input_reports() done, defined at /usr/local/lib/python3.7/dist-packages/joycontrol/server.py:17> exception=NotConnectedError(OSError(107, 'Transport endpoint is not connected'))>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/joycontrol/transport.py", line 153, in write
    await self._loop.sock_sendall(self._itr_sock, _bytes)
  File "uvloop/loop.pyx", line 2483, in sock_sendall
OSError: [Errno 107] Transport endpoint is not connected

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/joycontrol/server.py", line 20, in _send_empty_input_reports
    await transport.write(report)
  File "/usr/local/lib/python3.7/dist-packages/joycontrol/transport.py", line 157, in write
    raise NotConnectedError(err)
joycontrol.transport.NotConnectedError: [Errno 107] Transport endpoint is not connected
```

mateuspestana, 2021-04-10:
I'm having the same error here. Any solution?

ebisbe, 2021-04-11:
nope from my side at the moment. I found Brikwerk/nxbt project but it's also non working...

closed 2022-05-24 by ebisbe