mart1nro/joycontrol #76: Random 'Conn

ebisbe, 2020-08-04:
```

pvanderlinden, 2020-12-05:
this is usually when the bluetooth controller power is set to off

ebisbe, 2020-12-05:
> this is usually when the bluetooth controller power is set to off

What do you mean? I'm using a Raspberry pi. It connects and I can send commands to use the switch.

pvanderlinden, 2020-12-05:
Mmh then I have misinterpreted what you said. But I got those kind of messages when the Bluetooth adapter power was off (which I could switch on with bluetoothctl)

ebisbe, 2020-12-06:
> Mmh then I have misinterpreted what you said. But I got those kind of messages when the Bluetooth adapter power was off (which I could switch on with bluetoothctl)

@pvanderlinden maybe I did something wrong. What command did you execute to turn it on?

pvanderlinden, 2020-12-06:
From the command line I ran `bluetoothctl power on` (so before starting joycontrol)

closed 2022-05-24 by ebisbe