mart1nro/joycontrol #78: BlockingIOError: [Errno 115] Operation now in progress

nicognaW, 2020-08-18:
```
nico@ubuntu:~/joycontrol$ sudo python3 run_controller_cli.py PRO_CONTROLLER -r MY_SWITCH_BLT_ADDR
Traceback (most recent call last):
  File "run_controller_cli.py", line 327, in <module>
    loop.run_until_complete(
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "run_controller_cli.py", line 283, in _main
    transport, protocol = await create_hid_server(factory, reconnect_bt_addr=args.reconnect_bt_addr,
  File "/home/nico/joycontrol/joycontrol/server.py", line 111, in create_hid_server
    client_ctl.connect((reconnect_bt_addr, ctl_psm))
BlockingIOError: [Errno 115] Operation now in progress
```

The machine I use is a Ubuntu VM.

nicognaW, 2020-08-18:
I used another usb adapter instead, and it worked.

closed 2020-08-18 by nicognaW

ebisbe, 2020-08-31:
> I used another usb adapter instead, and it worked.

Mind you sharing which one are you using? My Pi drops the connection and I bougth a plugable.com usb one but out of the box is not working.