mart1nro/joycontrol #80: Request to Change License to Apache 2.0

YouTubePlays, 2020-09-02:
Hello,

@spacemeowx2 has decompiled JoyCon Droid and included transcribed code into this project. I think it is only fair if you change Licence of this project to Apache 2.0 so I don't seem to infract on your licence.

Best

YouTubePlays, 2020-09-02:
Or if you dont want to change the licence you can either grant me an exception or remove revert the mentioned contribution.

YouTubePlays, 2020-09-02:
[Transcribed Code Contribution](https://github.com/mart1nro/joycontrol/commit/be8dce71a0bcddd00b7b599bed3b8f495f7b87e8)

Proof:
[image]

spacemeowx2, 2020-09-02:
That's true. And I'm sorry what I did. 

I choice to copy NFC protocol data from the app instead of capturing data by myself. Because I wanted to go shortcut.

mart1nro, 2020-09-02:
@YouTubePlays 
I'm sorry this happened and have reverted the mentioned contribution from @spacemeowx2 

This means, that currently the nfc emulation support is now removed from joycontrol.

YouTubePlays, 2020-09-02:
@mart1nro if you are willing to give me an exception on your GPL 3.0 then I dont mind it is being included in this repository.

YouTubePlays, 2020-09-03:
Thanks. And Let me know if you change your mind about the subject above.

closed 2020-09-03 by YouTubePlays

choss, 2020-09-05:
@YouTubePlays: it doesn't work like that. 
The only thing what can help is if you (as the author of the original code) grant a GPLv3 licence to to this project for your code. Of course from this point on the implementations might drift a bit...
Everybody who submitted code to this repo here did it under GPLv3 and therefore in order to change the licence, one would have to ask every author.

corndog2000, 2021-01-31:
Is there anyway to get access for the code that has working NFC/Amiibo functionality? That is the only reason I found this repository and not the feature is no longer working.

andrewthenintendrone, 2021-02-28:
@corndog2000 You could always get the commit from right before the functionality was removed.
git reset --hard bf2e7e52fb54edb6b74e6a21fb77e0f73f14e21e

deltabeard, 2021-03-04:
> Apache 2 software can therefore be included in GPLv3 projects, because the GPLv3 license accepts our software into GPLv3 works.

https://www.apache.org/licenses/GPL-compatibility.html

My understanding is that there should therefore be no licensing issues here.

Codingale, 2021-03-22:
> https://www.apache.org/licenses/GPL-compatibility.html
> 
> My understanding is that there should therefore be no licensing issues here.
Came here to say this after I updated my version and saw this. The only real issue here would be if the reverse happens, which from what I can tell YouTubePlays wanted the code here, but didn't want to relicense their transpiled code, which is fine and means the project didn't need to revert the commit, if you wanted you could just add a notice of sub license to the top of the file.

Least reverting reverted commits works mostly fine.

DanTheMan827, 2021-04-08:
@YouTubePlays Why not just open source joycon droid?

Two groups working towards a common goal means faster discovery of fixes for everyone.

What reason is there to even keep it closed source?

DVDIsDead, 2022-04-27:
> What reason is there to even keep something like this closed source?

i second this