mart1nro/joycontrol #81: Possible to use wired controller?

IndiumCrypto, 2020-09-12:
I would like to know if it's possible to use a wired ps4 controller to connect to the script? I'm trying to play nintendo switch with a buddy with me over the internet.

SomeWhatHappyDuk, 2020-09-13:
Your best bet would probably be using pygame to capture your wired controller's inputs and then mirror those inputs and send it to the switch. I have done this with my wired Xbox 360 controller and it worked fine.

IndiumCrypto, 2020-09-13:
Could you post some kind of source code? I have 0 experience in Python an i just want to play a game with my mate across the world

SomeWhatHappyDuk, 2020-09-13:
Quick clarification, your friend is running the script right?

IndiumCrypto, 2020-09-13:
I'm running the script. But we're using parsec to passthrough his controller and screen capture

SomeWhatHappyDuk, 2020-09-13:
I made a just fork of Joycontrol with my Controller script, Now all you have to do is download it on your raspberry pi or whatever you're running the script on and run it, then type the command "Controller" and it should work. The button mapping might be all messed up because I made this script for an Xbox controller and I don't happen to have a wired PlayStation controller sitting around but if you run into any problems I would be happy to help you.

You can find the fork here:
[https://github.com/SomeWhatHappyDuk/joycontrol](https://github.com/SomeWhatHappyDuk/joycontrol)