mart1nro/joycontrol #82: Add NFC functionality (for amiibos mainly)

choss, 2020-09-13:
Since we needed to do remove nfc in context of #80 we should implement it again.

For this we need to implement MCU emulation in joycontrol. 
The relevant reading materials are
- https://gist.github.com/CTCaer/69ce224521d1ee4b153b250d9cbbb6a9
- https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/issues/49

The second link describes the the subcommands somewhat. Also the crc8 at the end of each response is mentioned in the thread.

A proposed design would be doing the MCU as a state machine and perhaps hardcoding the relevant initialization responses, read an arbitraty length tag (also a state in the machine) and also be able to change the state back to init/etc based on the subcomands.

Is somebody working on this already? 
I am no python expert but would like to help.

Kagenokami, 2020-09-22:
Any news on this? I downloaded this application solely for the NFC amiibo support, without this functionality the the time I spent getting it to work on my device was essentially wasted. :/

Poohl, 2020-11-05:
Hi, yes I'm working on it.
My current implementation is still rather hacky but is capable of simulating the tag good enough to have the switch read it successfully and afterwards return to regular input.
See https://github.com/Poohl/joycontrol/tree/amiibo_edits for the implementation.
The implementation doesn't really use a state-automata yet, since I haven't figured out enough transitions and is also barely tested.
I tested only using Splatoon2, the PRO_CONTROLLER setting, my laptop's (acer spin 5) builtin bluetooth, bluez5.53 (5.55 seemed to not work) and an Amiibo-dump I found online. This means I haven't yet verified that Amiibos with registered owners work, since I didn't yet manage to get mine dumped.
So if anyone wants to help, testing would be greatly appreciated and/or figuring out how writing to amiibos works.

Another great resource I found:
https://github.com/CTCaer/jc_toolkit/blob/5.2.0/jctool/jctool.cpp at c.a. line 2523

edwardsaunders7, 2020-12-08:
Any updates on NFC Support?

Poohl, 2020-12-09:
I just created a pull request. I got it working simulation a Pro controller. But since only read works I cannot yet verify it actually working in game (it just asks me to register a user)

choss, 2020-12-09:
Cool stuff! Let me grab your branch soon(tm) and test this out with some good ACNH

JobSoltero, 2020-12-11:
Ill be testing this weekend too, on skyrim.

corndog2000, 2021-01-31:
@JobSoltero How do you load an NFC file with your fork of the code?

Poohl, 2021-01-31:
> @JobSoltero How do you load an NFC file with your fork of the code?

either at startup add the  `--nfc <path_to_the_bin_dump>` option or in the terminal use `nfc <path_to_the_bin_dump>`  command.

Poohl, 2021-01-31:
Ok, general update: I tested BOTW, and since you don't have to register an owner there for some reason, it just works.

I also started on supporting writing amiibos and wrote a wireshark plugin to dissect them. If anyone wants to help dissecting captures from the proxys that write to amiibos: [NXBT wireshark Plugin](https://github.com/Poohl/Nintendo_Switch_Reverse_Engineering/tree/nxbt_wireshark_plugin/packet_parse/nxbt)

The "write too amiibo" command seems to be `nxbt.mcu.subc == 0x08`

corndog2000, 2021-01-31:
@Poohl Those commands don't seem to work. I get errors that send me back to github #80. Using the argument I get a bunch of errors saying: 
`[14:30:44] joycontrol.protocol input_report_mode_full::172 WARNING - NFC communictation is not implemented.`

Poohl, 2021-01-31:
which branch are you on? the changes are on amiibo_edits

KeesG, 2021-02-05:
Hey, is it possible to register the amiibo? for example in splatoon 2? I am using a random bin file i found right now and i was hoping you knew what to do.

Poohl, 2021-02-05:
I'm working on that. Not yet, sadly. For registering an Amiibo writing to them needs to be supported but whilst I am confident the core part works as I posted above in the gist, I assume that there is a lot more subtle stuff we havent't figured out. Also I didn't have time to test this yet.

KeesG, 2021-02-05:
I got it working after hours of messing around with different branches. But for me its working perfectly now, really quick and the nfc is great on acnh and works on botw & age of calamity too. Thanks for the work you put in :)

Poohl, 2021-02-20:
Update on writing: I can get the switch to transfer the entire write-request, but since I have no capture of what should happen next the transaction dies after that. I'm trying to get a capture of that using the proxy, but I had no luck yet.

Poohl, 2021-03-21:
Amazing news: It kinda works. I figured out how to edit the dump to make it be valid despite the filed write. The edits are on another branch: [amiibo_writing](https://github.com/Poohl/joycontrol/tree/amiibo_writing) (Use `git checkout amiibo_writing` after you have cloned the repo to get there)

I opened another Pull request as WIP for this, see #102 for instructions on how to use it.