mart1nro/joycontrol #87: Adding support fort left/right joystick push/press ?

mixaal, 2020-11-29:
Hi, 

I love this project since and I use it as a joycon replacement. The last thing I'd love to see is the left/right joystick press. It's used in some games for quite important things like "Lock on the powerful foe" in Hyrule Warriors .

choss, 2020-11-29:
How about the commands: 'r_stick' and 'l_stick' ?

mixaal, 2020-11-29:
Yes, I know they are there and use them to put the X,Y value, but can they do the press ?

mixaal, 2020-11-29:
Oh, maybe `queue.add('press r_stick')` ?

indykoning, 2020-12-19:
having made https://github.com/indykoning/joycon-controller
i can verify it has support for it, as you csan see in: https://github.com/mart1nro/joycontrol/blob/18a09da1a04306534ff9e1df8a1a69c0192a3244/joycontrol/controller_state.py#L81

The command would be press/push l_stick