mart1nro/joycontrol #93: Connection is immediately lost

syveqc, 2020-12-22:
Hi, I just tried to connect, and the switch seemingly recognizes a new controller, but as soon as I try to send a command, connection is lost:

[14:32:53] joycontrol.device set_name::69 INFO - setting device name to Pro Controller...
[14:32:53] joycontrol.server create_hid_server::83 INFO - Advertising the Bluetooth SDP record...
[14:32:53] joycontrol.device set_class::61 INFO - setting device class to 0x002508...
[14:32:53] joycontrol.utils run_system_command::75 DEBUG - ['hciconfig hci0 class 0x002508' exited with 0]
[14:32:53] joycontrol.server create_hid_server::96 INFO - Waiting for Switch to connect... Please open the "Change Grip/Order" menu.
[14:32:54] joycontrol.server create_hid_server::100 INFO - Accepted connection at psm 17 from ('DC:68:EB:13:DE:F6', 17)
[14:32:54] joycontrol.server create_hid_server::102 INFO - Accepted connection at psm 19 from ('DC:68:EB:13:DE:F6', 19)
[14:32:54] joycontrol.protocol connection_made::108 DEBUG - Connection established.
[14:32:56] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.REQUEST_DEVICE_INFO
cmd >> [14:32:56] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SET_SHIPMENT_STATE
[14:32:56] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SPI_FLASH_READ
[14:32:56] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SPI_FLASH_READ
[14:32:56] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SET_INPUT_REPORT_MODE
[14:32:56] joycontrol.protocol set_reader::361 INFO - Setting input report mode to 0x30...
[14:32:56] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.TRIGGER_BUTTONS_ELAPSED_TIME
[14:32:56] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SPI_FLASH_READ
[14:32:56] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SPI_FLASH_READ
[14:32:57] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SPI_FLASH_READ
[14:32:57] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SPI_FLASH_READ
[14:32:57] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SPI_FLASH_READ
[14:32:58] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.ENABLE_6AXIS_SENSOR
[14:32:58] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.ENABLE_VIBRATION
[14:32:58] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SET_NFC_IR_MCU_CONFIG
[14:32:58] root _reply_to_sub_command::244 INFO - received output report - Sub command SubCommand.SET_PLAYER_LIGHTS

cmd >> a
cmd >> [14:33:04] joycontrol.transport read::105 ERROR - No data received.
[14:33:04] joycontrol.protocol connection_lost::113 ERROR - Connection lost.

After that I get a NotConnectedError (which is to be expected I guess after losing connection):

cmd >> a
[14:33:12] __main__ _main::304 INFO - Stopping communication...
Traceback (most recent call last):
  File "run_controller_cli.py", line 329, in <module>
    _main(args)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "run_controller_cli.py", line 302, in _main
    await cli.run()
  File "/home/tobias/git/joycontrol/joycontrol/command_line_interface.py", line 197, in run
    await button_push(self.controller_state, *buttons_to_push)
  File "/home/tobias/git/joycontrol/joycontrol/controller_state.py", line 235, in button_push
    await button_press(controller_state, *buttons)
  File "/home/tobias/git/joycontrol/joycontrol/controller_state.py", line 206, in button_press
    await controller_state.send()
  File "/home/tobias/git/joycontrol/joycontrol/controller_state.py", line 66, in send
    await self._protocol.send_controller_state()
  File "/home/tobias/git/joycontrol/joycontrol/protocol.py", line 58, in send_controller_state
    raise NotConnectedError('Transport not registered.')
joycontrol.transport.NotConnectedError: Transport not registered.

I did the --noplugin=insert from #8. The readme states that I could also do this via USB, how would I accomplish that? If I disable bluetooth completely and connect my computer to the switch via USB, the script no longer works.

Thanks in advance!

syveqc, 2020-12-23:
I was able to get my hands on a Raspberry and it works there, so the problem is probably due to the bluetooth on my notebook! I would still be interested in how to accomplish this via USB.
Thanks in advance!

Poohl, 2021-03-15:
Quick question: What version of Switch os, raspbian and bluez are you running on the pi and on the Laptop what version of bluez? (run `apt-cache policy bluez` for bluez version, should be in the range 5.5X)

Also if you really want USB, there are a few projects:
- https://github.com/progmem/Switch-Fightstick for I believe a Teenzy or really anything capable of running LUFA
- https://github.com/shinyquagsire23/Switch-Fightstick for a Arduino Uno R3
- https://github.com/Poohl/gadfetfs-Switch-Fightstick for anything running Linux and a USB device controller
Note however these are not based on the Pro-Controller and thus have no NFC, Gyro, (maybe Rumble) etc...