mart1nro/joycontrol #95: USB controller

nico663, 2021-01-06:
Hey @mart1nro, is there a possibility to connect USB controllers (e.g. PS3) and use it with _joycontrol_ ?

Daniel2193, 2021-01-12:
Well in theory you can use any form of input device. I made my own script that uses asyncio_dgram to act as a udp "server". So i can use a C# application on my windows 10 pc to use my keyboard / xbox360 controller. With that I can allow others to connect to my pc via parsec so i can play stuff like hyrule warriors coop with them. This repo has everything you need to do that

moeiscool, 2021-01-27:
@nico663 I'm not seasoned in this stuff but if you are I suggest looking at this https://github.com/wchill/SwitchInputEmulator if you know your way around stuff. Just be sure to ask the author for permission if you intend to merge any of that code into joycontrol.

nico663, 2021-03-20:
> @nico663 I'm not seasoned in this stuff but if you are I suggest looking at this https://github.com/wchill/SwitchInputEmulator . 
Thank you @moeiscool

closed 2021-03-20 by nico663

moeiscool, 2021-03-29:
@nico663 I figured this one out with another repo (https://github.com/milador/RaspberryPi-Joystick). This method requires an OTG capable computer board (like Raspi 4 with USB-C).

I went as far as to build up a web interface engine for JoyControl and RaspberryPi-Joystick. I call this wrapper Kocho (it wraps both of those controller engines) https://gitlab.com/Shinobi-Systems/Kocho

I installed 3 JoyControl versions of Kocho onto a single machine without too many problems.
I installed 1 RaspberryPi-Joystick version on  its own machine, then use another machine for video processing (I'm sure I could make the Pi do it but it was easier to use another computer board for me due to power restraints).

Here is a catalogue of some of the usage that happens with my personal copies of Kocho. https://www.twitch.tv/moewashere I use an external script to monitor it and start recording when people come and play with my web ones.

@Daniel2193 this might interest you as well :) cheers!