mart1nro/joycontrol #98: Steam Controller Support

dqtech, 2021-01-26:
I would like to use this project to create an adapter to use a Steam Controller as a Nintendo Switch Pro controller wirelessly. I have seen many people have forked the project to include support for PS3, PS4 and even XBOX controllers. Unfortunately my python skills are limited so I don’t know where to start.

afonsopbarros, 2021-02-25:
I can't find the forks with support for XBOX/PS4 controllers, could you link me please?

dqtech, 2021-02-26:
@afonsopbarros 
Here are some links to the forks I have viewed:
1) Xbox One Wireless & Wired Controller 
https://github.com/Deltara3/joycontrol

2) Xbox Controller (not sure which one, but it's listed in the code)
https://github.com/SomeWhatHappyDuk/joycontrol

3) PS3, PS4 and Xbox360
https://github.com/diesel701/joycontrol/blob/controllers/use_controller.py

indykoning, 2021-03-20:
You could try to take a look at: https://github.com/indykoning/joycon-controller

It uses a generic controller input library so it may have more success, however you may have to do some configuring yourself