mart1nro/joycontrol #104: Unable to connect with Switch update 12.0.0

illie0, 2021-04-07:
I have two switches, and one of them has been upgraded to 12.0.0 and it is unable to connect through the joycontrol app. It hangs at "Waiting for Switch to connect". I've confirmed that it functions properly still with 11.0.1. 

[17:47:31] joycontrol.device set_name::69 INFO - setting device name to Pro Controller...
[17:47:31] joycontrol.server create_hid_server::83 INFO - Advertising the Bluetooth SDP record...
[17:47:31] joycontrol.device set_class::61 INFO - setting device class to 0x002508...
[17:47:31] joycontrol.utils run_system_command::75 DEBUG - ['hciconfig hci0 class 0x002508' exited with 0]
[17:47:31] joycontrol.server create_hid_server::96 INFO - Waiting for Switch to connect... Please open the "Change Grip/Order" menu.

illie0, 2021-04-07:
This doesn't seem to be specific to joycontrol, as I see tweets about the android app "JoyCon Droid" also not working for people who upgraded to 12.0. So something has changed in the controller pairing.

RocMax, 2021-04-07:
Same on my two switches, unfortunately I updated both of them :(

IRC-Elmak, 2021-04-07:
also having the same issues.  I am regretting updating my switch to 12.00

sir-joel, 2021-04-07:
I'm encountering this problem when trying to connect from a new computer, but a computer that I previously used, which has a .reconnect file present, can still connect without issue.

From the new computer, however, I've noticed that if you let the Switch sit on the Change Grip/Order screen long enough, the system will eventually crash, stating that it has encountered and error and needs to restart. This happens consistently.

Nebulas123, 2021-04-08:
> Same on my two switches, unfortunately I updated both of them :(
My switch was automatically updated to 12.0.0, and the bluetooth just disconnects and connects repeatedly when I try to use joycontrol.....

dentedghost, 2021-04-08:
DeepBlue LabsApril 6, 2021
We noticed the JoyCon Droid is not compatible with 12.0.0 or above. We are working on the issue but it might take time.

Sent from my iPhone

> On Apr 6, 2021, at 6:07 PM, illie0 ***@***.***> wrote:
> 
> JoyCon Droid

dentedghost, 2021-04-08:
In 12.0 switch did update the Bluetooth-sysmodule

Sent from my iPhone

> On Apr 7, 2021, at 10:20 PM, S C M ***@***.***> wrote:
> 
> ﻿
> 
> DeepBlue LabsApril 6, 2021
> We noticed the JoyCon Droid is not compatible with 12.0.0 or above. We are working on the issue but it might take time.
> 
> Sent from my iPhone
> 
>>> On Apr 6, 2021, at 6:07 PM, illie0 ***@***.***> wrote:
>>> 
>> JoyCon Droid

tomifil, 2021-04-09:
I can confirm is not working anymore as switch update 12.0

Brikwerk, 2021-04-11:
Hi all,
I work on NXBT, a similar project to Joycontrol, and we've also been dealing with comparable issues after the v12.0.0 update. After some changes/fixes were applied to NXBT, it was able to pair/connect and communicate with the Switch (on v12.0.0) again, so controller emulation is still possible with Linux and BlueZ. I found that the main changes that come with the v12.0.0 update are:

- The Switch is pickier with which controllers it connects to. More aspects of the authentic Pro Controller need to be emulated to lure the Switch into connecting.
- The connection procedure has changed slightly when first communicating with the Switch.
- The Switch will kick controllers off that exceed a certain input frequency (typically 15Hz) in certain scenarios. I'm still looking into the exact mechanism by which the frequency is updated.

For those interested in further details on the changes, you can refer to [this issue](https://github.com/Brikwerk/nxbt/issues/18#issuecomment-817268531) which contains a more detailed writeup.

If I find some time, I'll see if I can port the changes needed over to this project and post a PR. NXBT uses a different emulation and protocol backend than Joycontrol so, unfortunately, it's not a 1-to-1 operation.

dentedghost, 2021-04-11:
I updated the [Ticket](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/issues/123) on Nintendo_Switch_Reverse_Engineering with @Brikwerk details and this thread.

illie0, 2021-04-13:
@pokemon-bot 
I don't believe that alone would be sufficient, as joycontrol tries to do some of this already, and in the wrong order or with values that need to be changed. For example it sets the class, and then sets discoverable on which causes the class to be reset to a different value. So some code changes are required.

Additionally, some bluetooth adapters may not be compatible now, such as the Qualcomm Atheros. I haven't found a way of changing the MAC address on them (they weren't supported by bdaddr, and doesn't use 'hcitool cmd 0x3f 0x001' either.)

thxomas, 2021-04-16:
I managed to get a running setup, with quite stable connection.

If interested, please refer to [https://github.com/Poohl/joycontrol/issues/3#issuecomment-821082464](https://github.com/Poohl/joycontrol/issues/3#issuecomment-821082464)

T.

thxomas, 2021-04-18:
Guys, I'm in hell... I cannot have the Switch to pair anymore since yesterday :(  I use the v12_fixes branch now, everything should be fine (MAC Addr., Name, SDP, simple-agent, Device Class, unpaired, /var/lib/bluetooth wiped,  ...) but it is not... I'm back at the start of this thread : "User Auth accepted", then "Link Key Exchange" then stall.
I can see these events in btmon, but no connection request ever reaches joycontrol socket.. 
Are you still having success on your side ?

Venryx, 2021-04-19:
@pokemon-bot The NXBT v12 branch worked fine for me, in an Xubuntu VM on Windows, with the ZEXMTE Bluetooth dongle.

Some more info here: https://github.com/Brikwerk/nxbt/issues/18#issuecomment-822151271

I tried using the joycontrol [v12 branch](https://github.com/Poohl/joycontrol/tree/V12_fixes), but I was unable to get it to work (the Pair popup just never showed up).

There were some limitations with NXBT (the fake-pro-controller would disconnect whenever I connected more than 1 physical joycon), but since it works for the most part, I'm sticking with that until joycontrol and/or NXBT get a more robust solution figured out.

Venryx, 2021-04-24:
Thanks for adding your experience @pokemon-bot, and great to hear that someone else has had success with joycontrol on 12.0+.

Regarding it working without the mac address being changed, this may be relevant (and the comment after it): https://github.com/Brikwerk/nxbt/issues/18#issuecomment-825877288

formatcom, 2021-09-16:
> If I were to add a connectWithRetry( maxRetry) call into joycontrol with the idea to connect then send an "a" button then wait to make sure it stays connected. If it disconnects. It will loop upto [...]

here https://github.com/Poohl/joycontrol/issues/11#issuecomment-920226012