mart1nro/joycontrol #112: Can't get to work on Pi Zero W

hungtr97, 2021-07-27:
I'm get log to this
```
pi@raspberrypi:~/joycontrol $ sudo python3 ./run_controller_cli.py PRO_CONTROLLER
[06:42:17] joycontrol.device set_name::69 INFO - setting device name to Pro Controller...
[06:42:17] joycontrol.server create_hid_server::83 INFO - Advertising the Bluetooth SDP record...
[06:42:17] joycontrol.device set_class::61 INFO - setting device class to 0x002508...
[06:42:17] joycontrol.utils run_system_command::75 DEBUG - ['hciconfig hci0 class 0x002508' exited with 0]
[06:42:18] joycontrol.server create_hid_server::96 INFO - Waiting for Switch to connect... Please open the "Change Grip/Order" menu.
```

I have disconnect real joycon and didn't touch it, but the virtrual controller didn't show up.

Please help,
Thanks.

oyylumo, 2021-09-25:
I'm getting the same issue with identical console output on switch fw 13.0.0

juancroldan, 2022-02-28:
Same messages in Raspberry Pi 4B