mart1nro/joycontrol #115: not quite sure what the issue is but i'll leave the error code here (running ubuntu 20.04 under windows WSL service)

suspect-407, 2021-11-21:
Traceback (most recent call last):
  File "run_controller_cli.py", line 328, in <module>
    loop.run_until_complete(
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "run_controller_cli.py", line 284, in _main
    transport, protocol = await create_hid_server(factory, reconnect_bt_addr=args.reconnect_bt_addr,
  File "/home/vince/joycontrol/joycontrol/server.py", line 45, in create_hid_server
    ctl_sock = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_SEQPACKET, socket.BTPROTO_L2CAP)
  File "/usr/lib/python3.8/socket.py", line 231, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 97] Address family not supported by protocol

RENOxDECEPTION, 2021-12-03:
I have this error as well.

KokoseiJ, 2021-12-16:
It is generally a bad idea to use a low-level hardware dependent program on WSL2. It's just a thin wrapper over Windows, and most of the APIs directly speaking to hardwares are not implemented.
If running Linux on a bare metal is not an option, try VirtualBox. VMWare uses its own Bluetooth wrapping layer rather than passing the device directly through so you might wanna avoid it.

kinhoy, 2022-03-15:
It seem to like that we can use https://github.com/dorssel/usbipd-win this project to make bluetooth working on WSL2, but I don't know how to use it.

virgil724, 2023-01-13:
[usbipd-win #310](https://github.com/dorssel/usbipd-win/discussions/310?sort=top#discussioncomment-2375886)

I got bluetooth work in WSL2 work well with nxbt 
with this repo i cannot got pairing succeed 
pair succeed with https://github.com/Poohl/joycontrol