mart1nro/joycontrol #118: stopped with HID attribute Error

giowoo, 2022-04-21:
t@t-VirtualBox:~/joycontrol$ sudo python3 ./run_controller_cli.py PRO_CONTROLLER
Traceback (most recent call last):
  File "./run_controller_cli.py", line 10, in <module>
    from joycontrol import logging_default as log, utils
  File "/home/t/joycontrol/joycontrol/utils.py", line 5, in <module>
    import hid
  File "/usr/local/lib/python3.8/dist-packages/hid/__init__.py", line 83, in <module>
    hidapi.hid_get_input_report.argtypes = [ctypes.c_void_p, ctypes.c_char_p, ctypes.c_size_t]
  File "/usr/lib/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /lib/x86_64-linux-gnu/libhidapi-hidraw.so.0: undefined symbol: hid_get_input_report

julianhu1979, 2022-04-23:
same issue. on Raspberry Pi 4

xhw0525, 2022-04-25:
pip install hid==1.0.4

xhw0525, 2022-04-25:
I can run normally But I can't connect to my switch

rorokapur, 2022-05-08:
I am having the same issue on a Pi 2