mart1nro/joycontrol #121: Does not connect to switch fw 14

darrenchang, 2022-05-26:
When I go to "Change Grip/Order" menu, the virtual pro controller never shows up. I've waited for 5 minutes and more, and still nothing shows up.
Is it because my Switch fw version is updated to 14?

- System: Ubuntu 22.04
- Python: 3.10.4
- Switch version: 14.1.1

```
/home/ubuntu0/git/joycontrol/run_controller_cli.py:327: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
[13:07:45] joycontrol.device set_name::69 INFO - setting device name to Pro Controller...
[13:07:45] joycontrol.server create_hid_server::83 INFO - Advertising the Bluetooth SDP record...
[13:07:45] joycontrol.device set_class::61 INFO - setting device class to 0x002508...
[13:07:45] joycontrol.utils run_system_command::75 DEBUG - ['hciconfig hci0 class 0x002508' exited with 0]
[13:07:45] joycontrol.server create_hid_server::96 INFO - Waiting for Switch to connect... Please open the "Change Grip/Order" menu.
```

darrenchang, 2022-05-26:
I removed the bluetooth device from my Ubuntu 22.04's setting app and it starts working. I can now connect to my switch as PRO_CONTROLLER.

closed 2022-05-26 by darrenchang

huangdihd, 2023-01-15:
I have this question too

huangdihd, 2023-01-15:
You can try using the following version of python3.10 to solve the problem,after 3.10, this method is no longer named by that name

huangdihd, 2023-01-15:
I'm testing it