mart1nro/joycontrol #130: Trouble installing

MagnificentCreature, 2023-03-06:
Hi so I'm downloading this in hopes to make a pokemon cloning macro before the event expires.

The issue is that I use a windows computer, so I'm downloading WSL to try and get this to function. I downloaded the WSL, and was able to run the sudo apt install python3-dbus libhidapi-hidraw0 lines after installing a few other things.

But I am unable to get the Archlinux derivatives nor have any clue how to get it done. Some assistance in installation would be very much appreciated ❤️ !

MagnificentCreature, 2023-03-06:
Update: I decided I might use VirtualBox instead of WSL

MagnificentCreature, 2023-03-06:
To elaborate I can’t run the pacman or makepkg command to download the archlinux dependencies because i don’t think WSL comes autoinstalled with them, and any guide on that would be much appreciated. Also I don’t know how to activate bluetooth on WSL (without a bluetooth dongle)

Beedit, 2023-04-24:
Hey this is probably very late and you might not need this anymore but just in case someone else runs into this issue.

You dont need to install the arch linux derivatives as they are not for the WSL you are using. They're for arch linux instead. pacman and makepkg are not installed on your WSL and they dont need to be. 

Hope this helps someone.